robotology/icub-firmware #488: Create a 2foc version for temperature testing/demo

valegagge, 2024-03-29:
For demo/testing purposes we want to develop a 2foc version that doesn't raise an HW fault when an overheating occurs (both in case the temperature goes beyond the hw limit and the fw gets an error reading for more than 10 seconds.)

Dod
Fw developed and works as expected

valegagge, 2024-03-29:
I created the required fw with version 3.3.27
EXE is here: https://github.com/valegagge/icub-firmware-build/tree/noFaultOverheating
SRC is here: https://github.com/valegagge/icub-firmware/tree/noFaultOverheating

I could not verify the fw.

cc @S-Dafarra @maggia80 @jacopo

valegagge, 2024-04-04:
FW updated on the robot and tested it.
CC @MSECode

valegagge, 2024-04-12:
FW run on robot. Closing

closed 2024-04-12 by valegagge

S-Dafarra, 2024-04-12:
Hi @valegagge, at the moment the situation is not ideal. It would be better to reenable the temperature check now, but in less than a month we will have another demo and most likely we will need again the same thing. How can we proceed?