robotology/icub-firmware #497: Too many debug messages on iCubGenova11

marcoaccame, 2024-05-29:
In doing the tests for this PR:
- https://github.com/robotology/icub-firmware/pull/496
 
I noticed that on `iCubGenova11` there are too many diagnostics debug messages. See picture.

[image]

I analyzed the log a bit and I found out that each 2foc board emit one periodic debug message of type `PER_MC_MSG__DEBUG` every 500ms + something. See in here the [code](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/embobj/plus/can/config-can-protocol/EoCANprotMCperiodic.c#L105-L136) that receives from the 2foc and sends the diagnostics message to YRI.

Talking to @MSECode, it may be that the 2foc emits because of some xml setting in [here](https://github.com/robotology/robots-configuration/blob/master/iCubGenova11/hardware/mechanicals/left_arm-eb1-j0_3-mec.xml#L29-L37), maybe the verbose setting that does not appears and is 1 by default. See [here](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjMotionControl/eomcParser.cpp#L1472).

cc @valegagge @ale-git @martinaxgloria @Nicogene

marcoaccame, 2024-05-29:
In analyzing the code of the 2foc w/ @MSECode we spotted a possible bug in [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/can_icubProto_trasmitter.c#L104-L127).
The transmitted `ICUBCANPROTO_PER_MC_MSG__DEBUG` message contain a dirty content in the two least significant bytes. In particular these two bytes contain the current emitted by the latest `ICUBCANPROTO_PER_MC_MSG__2FOC`. See [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/can_icubProto_trasmitter.c#L72).

The above is coherent with the diagnostics messages that shows all zero apart the values of the two LSB:

[image]

valegagge, 2024-05-29:
Hi @marcoaccame and @MSECode ,
good catch!
yes, the 2foc fw sends a packet with length =8, but it fills 6 bytes only.

Of course, we can fix it, but, in any case, the information put in the DEBUG message is no longer useful.

Doing another work related to the study of quadrature encoder, I'm evaluating to use that message for sending encoder raw values.

So, in the end, I suggest fixing it by setting the default value of the verbose parameter equal to 0. It should be set to 1 only when the autocalibration is performed or in debug session by technical people.

What do you think?

marcoaccame, 2024-05-29:
> So, in the end, I suggest fixing it by setting the default value of the verbose parameter equal to 0. It should be set to 1 only when the autocalibration is performed or in debug session by technical people.

I tested again on the robot and the messages go away if I explicitly set verbose = 0. So: yes we should change the default value of `Verbose` to 0 so we don't have to modify the xml files of all robots.

> Doing another work related to the study of quadrature encoder, I'm evaluating to use that message for sending encoder raw values.

You mean that the 2foc transmits the qenc value with the `ICUBCANPROTO_PER_MC_MSG__DEBUG`? It is fine by me. Or maybe you can use a new dedicated message able to transport qenc values but also other values depending on a param for instance in byte[0]. As you like.

MSECode, 2024-05-29:
> I tested again on the robot and the messages go away if I explicitly set verbose = 0. So: yes we should change the default value of `Verbose` to 0 so we don't have to modify the xml files of all robots.

That's good. So it was actually due to that as thought earlier. 
It make sense to modify icub-main and set the `verbose` `default` value to `zero`.

valegagge, 2024-05-29:
> You mean that the 2foc transmits the qenc value with the ICUBCANPROTO_PER_MC_MSG__DEBUG? 

Yes!

> It is fine by me. Or maybe you can use a new dedicated message able to transport qenc values but also other values depending on a param for instance in byte[0]. As you like.

Exactly... I'm evaluating this alternative option also!

MSECode, 2024-05-30:
I've done a first change setting the default value for verbose to 0 in [here](https://github.com/MSECode/icub-main/commit/a5c13e7451d0877a6eddc0eef796e5511d35aebc).
I suppose that for solving this bug, changing icub-main should be fine as pointed out in here: 
> I tested again on the robot and the messages go away if I explicitly set verbose = 0. So: yes we should change the default value of `Verbose` to 0 so we don't have to modify the xml files of all robots.

Let's see uf there is something else to modify on other related repos, such as the proposal in here: 
> You mean that the 2foc transmits the qenc value with the `ICUBCANPROTO_PER_MC_MSG__DEBUG`? It is fine by me. Or maybe you can use a new dedicated message able to transport qenc values but also other values depending on a param for instance in byte[0]. As you like.

If it's ok to modify only the icub-main part relatively to this bug I'll proceed with testing phase.

valegagge, 2024-05-30:
> If it's ok to modify only the icub-main part relatively to this bug I'll proceed with testing phase.

Yes, please proceed with the testing phase.

closed 2024-05-31 by pattacini