robotology/icub-firmware #499: Improve the rotor/stator calibration in 2FOC

valegagge, 2024-06-07:
After some tested carried on in https://github.com/icub-tech-iit/ergocub2-design-upperbody/issues/43, we noticed that sometimes joint couldn' t complete the calibration because it doesn't find the index before the timeout of 30 seconds exipre. 

We noticed that the motor does 3 or 4 complete tourns but the index interrupt isn't t triggered. So we need to create a setup with the oscilloscope to verify the index signal.

Moreover, in the procedure, some time constants need to be updated (or parametrized) since the 2foc now runs at 20KHz and not 40 KHz.

For more details please see [this comment](https://github.com/icub-tech-iit/ergocub2-design-upperbody/issues/43#issuecomment-2154436618).

**Dod**
The new procedure runs correctly both on the robot and on ergocub2 joints

MSECode, 2024-06-13:
After some knowledge transfer done with @ale-git, we understood that some changes should be done to the heuristic use in the method of the 2FOC where we perform the stator-rotor calibration looking for the index of the quadrature encoder mounted at the motor side of the joint.

The whole portion of code we are interested in is at this lines: https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L353-L520

Here a brief explanation of what should we modified and what probably was the cause of the failing of the calibration procedure.

- First of all we need to clarify that if we do not want to find the `stator-rotor offset`, so we are just looking for the index of the encoder and setting the zero at that position, we are just passing through the steps of the state machine defined by values `0-1-2` of the variable `sAlignInProgress`. So in this case we can limit our analysis to the lines [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L353-L520).
- Considering this we can see a couple of variables that we need to take care of. They are: 
    - limit in the rotation of the fast shaft when looking for the index defined by [this one](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L363)
    - timer/counter used for moving the fast shaft with a `stairs-like` curve, which defined here: https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L383 and used in the if clauses with this value: https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L391
- Supposed all of that we have to clarify that even if in different robots and setups we are using the same fw the hardware instead changes. Therefore, using a fw with values that are not parametrized depending on the multiple configurations we have it is most likely the cause of the misbehaviors we saw.
- To make an example we can talk about 2 different configurations:
    - ergoCub robot is using on the legs motors with 4 polar couples 
    - in our setup the new joints have motors with 10 polar couples
    - considering this, the value defined [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L363), which is based on the electrical degrees per revolution (= (motor poles / 2) * 360 ), changes a lot when using different hw configurations. This can be a problem of the calibration since we are using that value for defining how many ticks should the encoder span on one direction and (eventually) on the other while looking for the index.
    - moreover, considering the counter used for moving the shaft step-by-step, we did not change the value when we decided to reduce the frequency of the 2foc loop from 40 kHz to 20 kHz. Therefore, since each call of the `alignRotor` method occurs in the DMA interrupt service routine, we are basically moving the fast shaft with half of the speed.
- on top of this, since the timeout of the calibration defined by the EMS is still of 30 seconds, it happens that in some occasion, especially when close to the hw limit, the encoder does not have enough time for finding the index since the shaft is too slow to move towards one direction and the other. As a matter of fact, when the joint remained in the `not calibrated` state we saw from the raw data that:
    - first thing first the raw value of the ticks spanned was way lower than the limit defined [here](https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L363)
    - and the state machine stopped at the first phase (sAlignInProgress variable fixed to 1). So the shaft only had time to move in one direction, which is probably the one towards the hw limit where it started. 

In conclusion, I did some tests by reducing the value of the timer for our setup where the number of motor poles is more than the double with respect to the motor in the robot and I was able to always finding the index.
Thus, we need to revise together the heuristic used and parametrized the timing so that we are good in all hw configurations

cc: @valegagge @ale-git

valegagge, 2024-06-13:
Hi @MSECode , thanks for the explanation and tests.

Which value of the timer did you use in the test on the ergocub2 joints?

Moreover, I suggest to go on in two steps:
1. double the velocity according to the new 2foc frequency and test it on ergocub.
2. find a first attempt of "formula" that parameterizes the number of polar couples and the 2foc frequency.

what di you think?

MSECode, 2024-06-13:
On our joints I just changed the value of the timer to 50. In this manner I should be able to complete the full state machine calibration in 27 seconds. Considering that if I need to complete the 1,5 revolution to one side and to the other with joints with 3600 electrical deg per revolution and considering that the 2foc loop is running at 50 us. We have that, with 50 cycles per loop, we can make the 10800 electrical deg in 27 seconds. 
So I'll try with this value, which should be more than enough for the robot too and then we can parametrized it to make it scalable with respect to the number of motor poles different joints have. 
So the step you have defining are good.

MSECode, 2024-07-11:
I'm gonna re-check it since I've done some tests but that was some weeks ago

valegagge, 2024-07-29:
PR merged. Closing

closed 2024-07-29 by valegagge