robotology/icub-firmware #517: Update the CAN protocol documentation regarding the battery messages

valegagge, 2024-09-30:
We discovered during the test that we wrote the CAN message data regarding the battery in wrong order, so we need to fix it.

In addition, we need to update the document with the new messages

**Dod**

The document is updated

MSECode, 2024-09-30:
Started to update the protocol docs at this branch: https://github.com/MSECode/icub-firmware/tree/feature/updateBatterProtocolDoc

valegagge, 2024-10-24:
Hi @MSECode ,
in the document we need to add:
- `BMS_STATUS_INFO`
- `BAT_STATUS_INFO`
- `ADVANCED_STATUS`
- `LOG`.
We need to explain that class 6 is also used for the polling message regarding the log. 

Please refer to [this code](https://github.com/valegagge/icub-firmware/blob/ed145373875f5b4d02307803ed91b7523f974421/emBODY/eBcode/arch-arm/board/ems004/appl/v2/src/eoappservices/embot_app_eth_theBATservice.cpp#L539) and the Biamino document that you find in attach, but take in account that the document contains two error:
 - the message packInfo has the wrong format
 - how the BMS sends the periodic messages

MSECode, 2024-11-04:
> Hi [@MSECode](https://github.com/MSECode) , in the document we need to add:
>
> * `BMS_STATUS_INFO`
> * `BAT_STATUS_INFO`
> * `ADVANCED_STATUS`
> * `LOG`.
>   We need to explain that class 6 is also used for [...]

This info have been added to the document. I've also detailed the content of the data that we have in those CAN frames.

valegagge, 2024-11-04:
Hi @MSECode ,
the doc is ready to be reviewed?

MSECode, 2024-11-05:
> Hi [@MSECode](https://github.com/MSECode) , Is the document ready to be reviewed?

I think so, I'll double check that. One thing that I need to conclude the documentation is having the wingst version for the new BMS so that I can distinguished that one from the old one that has already a version.

valegagge, 2024-11-05:
ok, please let me know when it is ready for the review. Regarding the version of BMS, I'll take take to get it.. For now you can use a placeholder.

thanks

MSECode, 2024-11-06:
This is the PR I've opened for the update of the documentation:

https://github.com/robotology/icub-firmware/pull/532

MSECode, 2024-12-02:
Pr merged. Closing.