robotology/icub-firmware #534: Study problem related to how joint trajectory is defined when calibrating with calibration type 10

MSECode, 2024-11-25:
As shown in this 2 comments:
- https://github.com/robotology/icub-firmware/issues/529#issuecomment-2497126733
- https://github.com/robotology/icub-firmware/issues/529#issuecomment-2486283511

we need to understand the "possible problem" related to how the joint trajectory is defined when using calibration type 10 for calibrating the joint at the hard stop.
One thing to mention is that, from the data we have dumped we have identified this behaviour as a possible problem, but we first need to better analyze that to be sure that it is actually a problem and why it is happening.

cc: @valegagge

valegagge, 2024-11-29:
Checking the code I noticed in the function `JointSet_do_wait_calibration` , the embeddeed motion control performs a `JointSet_do_odometry` and then set the joint in `position control mode` when the hardstop is reached. I guess these two operations are the cause of the change of the joint position and of the trajectory.

@MSECode Can you check and confirm, please?

MSECode, 2024-11-29:
> Checking the code I noticed in the function `JointSet_do_wait_calibration` , the embeddeed motion control performs a `JointSet_do_odometry` and then set the joint in `position control mode` when the [...]

Yes exactly.

valegagge, 2024-11-29:
ok so we can close this issue.

closed 2024-12-03 by MSECode