robotology/icub-firmware #535: Analyze problem of AMO reading random joint position values

MSECode, 2024-11-26:
As already noticed in some occasions and mentioned in the following issues:
- https://github.com/robotology/icub-firmware/issues/428
- https://github.com/icub-tech-iit/study-encoders/issues/31
- https://github.com/icub-tech-iit/study-encoders/issues/22#issuecomment-2464924451

we have seen that it might happen that the AMO encoder starts, at a certain point, to read values that are not compliant with the expected ones given by the imposed movement. After some tests, most of which shown in this issue: https://github.com/icub-tech-iit/study-encoders/issues/31, we have observed that this problem verifies deterministically when we move the AMO sensor away from the magnetic disk while it is moving. As a matter of fact, in that configuration, i.e. when the receiver and the magnetic disk are not in the standard location (which should be extremally accurate and with really minimal variations from what written in the documentation), the encoder board is not able anymore to make the transformation mandatory for writing in the registries a meaningful value of the raw position, so that the controller can rescale that to degree inside its working loop.
Therefore, in order to reproduce this behavior, I've moved the encoder in velocity mode quite slowly, thus to add to the reading less disturbances as possible, and then I start to take the receiver far away from the magnetic disk.

Dod

Verifies the behavor documented in the issues and seen on the robot where we start to have "random" readings

MSECode, 2024-12-03:
**NOTE: THIS IS TO BE ADDRESS BY DISCUSSION ABOUT HOW TO MANAGED THE DIAGNOSTIC DATA COMING FROM THE HAL**

valegagge, 2024-12-20:
Hi @MSECode ,
we can close this issue in favor of a new one where we'll check if it is possible can AMO encoder can move outside its housing. https://github.com/icub-tech-iit/study-encoders/issues/37

closed 2024-12-20 by valegagge

MSECode, 2024-12-20:
Sure, moreover I'll keep in mind that we still need to open an issue to address the problem of the improvement of the low level (HAL) diagnostic of the AMO. Do we prefer to keep this discussion on a separate issue or make it part or continuation of https://github.com/icub-tech-iit/study-encoders/issues/37 ? What do u think?

valegagge, 2024-12-30:
Thanks for pointing out this. I suggest opening a new issue in icub-fw and adding it to the epic https://github.com/icub-tech-iit/study-encoders/issues/21. Thanks