robotology/icub-firmware #537: Test on the robot of new features and bugfix for calibration type 10 for AMO encoder

MSECode, 2024-11-26:
In this issue we are going to add the outcomes of the test on the robot for the enhancements and bugfixes done to the calibration type 10 process.

MSECode, 2024-11-28:
### Tests have been done on `ergoCubSN000`
First of all I'd like to report the status I found the robot, so that if any error happens we can recover from the repos I've rebased:

- superbuild
[image]

- icub_firmware_shared
[image]

- ICUB
[image]

- YCM
[image]

- YARP
[image]
 
- icub-firmware-build
[image]

#### First step updating of the `robotology-superbuild`

Everything oks. I moved the repos as follows:
- robotology-superbuild --> pulled from latest master
- icub-firmware-shared --> pulled latest devel
- ICUB --> pulled latest devel
- YCM --> last tag v0.17.1
- YARP --> last tag v3.10.1

valegagge, 2024-11-29:
Hi @MSECode, 
well done! Thanks for also reporting the previous state of the robot and the state after the update. 🚀 Since you didn't get errors to can proceed to create the PRs.

MSECode, 2024-12-02:
PRs opened and linked all to this one: https://github.com/robotology/icub-firmware-build/pull/174

closed 2024-12-20 by MSECode