robotology/icub-firmware #541: Analyze problem of AMO readings when a cable is broken/partyially broken or detached

MSECode, 2024-12-03:
Here we are going to analyze the second problem mentioned here: https://github.com/icub-tech-iit/study-encoders/issues/22#issuecomment-2464924451, which is the issue of understanding how the AMO encoder is reading the signals if the cable is not detached but just partially broken.
In order to simulate that we just removed one of the data cables from the AMO connector.

MSECode, 2024-12-03:
If we removed just one of the data cables of the AMO encoder connector we can see that the joint, if using the AMO encoder, is not able to calibrate as shown by the logs below.
Moreover, as shown by the plot we do not get any errors. As a matter of fact, we just reading zeros for both data and diagnostic.

### AMO raw readings

[image]

### Joint positions

[image]

### Logs

[log-yri-amo-cable-broken.log](https://github.com/user-attachments/files/17990396/log-yri-amo-cable-broken.log)

cc: @valegagge

MSECode, 2024-12-03:
**NOTE: TO BE EVENTUALLY DISCUSSION --> IMPROVE DIAGNOSTIC????**

valegagge, 2025-01-14:
Hi @MSECode please add a comment.

MSECode, 2025-01-14:
During the tests we have observed that if either the AMO encoder cable is broken/detached or partially broken, i.e. one of the signal cables is detached from the connector, the joint is always moved in HW FAULT CONDITION.

closed 2025-01-14 by MSECode