robotology/icub-firmware #547: Improve the Aksim diagnostic when the aksim cable is disconnected

valegagge, 2024-12-12:
During some tests performed on three-encoder setup, we noticed that when we disconnect the cable the reported diagnostic refers to other error types:

`[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=5125s 462m 295u :  HW - encoder: not connected (Joint=three_encoders_joint (NIB=0) Number of error in 10 seconds is: 2383)
[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=5125s 462m 400u :  HW - encoder: the position data is not valid (Joint=three_encoders_joint (NIB=0) Number of error in 10 seconds is: 2383)
[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=5125s 462m 527u :  HW - encoder: the position data is valid, but some operating conditions are close to limits. (Joint=three_encoders_joint (NIB=0) Number of error in 10 seconds is: 2415)
[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=5125s 462m 631u :  HW - encoder: inverted CRC is invalid (Joint=three_encoders_joint (NIB=0) Number of error in 10 seconds is: 2431)`

It seems that the FW signals all errors together because they are at the same millisecond. 

**Dod**
We want to notify only the error "encoder not connected".

PR with the fix done.

AntonioViscomi, 2024-12-16:
## WIP

Hi, I have done some testing.
I put the error handling component for the AksIM-2 encoder (which was already written and tested) inside an if/else statement to act, broadly, as follows:
 1. Checks if a HW error has occurred. If so, the error manager is responsible for raising the appropriate error message.
 2. If the condition is false, the error manager takes care of any other error condition (invalid CRC, invalid data, close to limits warning).

The errors are then visualized every 10 seconds.

If the encoder connector is disconnected, the output of the `yarprobotinterface' is as follows:

```
[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=267s 275m 291u :  MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=three_encoders_joint (NIB=0), encoderPort=0)

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=272s 276m 291u :  MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=three_encoders_joint (NIB=0), encoderPort=0)

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=275s 746m 214u :  HW - encoder: not connected (Joint=three_encoders_joint (NIB=0) Number of error in 10 seconds is: 10001)

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=277s 277m 291u :  MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=three_encoders_joint (NIB=0), encoderPort=0)

[DEBUG] yarprobotinterface running happily

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=282s 278m 291u :  MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=three_encoders_joint (NIB=0), encoderPort=0)

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=285s 747m 214u :  HW - encoder: not connected (Joint=three_encoders_joint (NIB=0) Number of error in 10 seconds is: 10001)

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=287s 279m 291u :  MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=three_encoders_joint (NIB=0), encoderPort=0)

[ERROR] from BOARD 10.0.1.1 (three-encoders-setup) time=292s 280m 291u :  MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. (Joint=three_encoders_joint (NIB=0), encoderPort=0)
```

While the `HW encoder: not connected` component seems to work as expected (if the callback runs at 1 k[Hz], 10001 errors are thrown in 10 seconds), there is a debug message concerning the AEA encoder, and at the moment I can't understand if this message is somewhat related to the AksIM.

valegagge, 2024-12-16:
Good work @AntonioViscomi !

I guess the error you mentioned above is related to some checks performed by the embedded motion control. Tomorrow we can have a look together!

Please put here the link to the branch of your fix. Thanks

AntonioViscomi, 2024-12-17:
> I guess the error you mentioned above is related to some checks performed by the embedded motion control. Tomorrow we can have a look together!
> 
> Please put here the link to the branch of your fix. Thanks

Hi @valegagge , [here](https://github.com/AntonioViscomi/icub-firmware/tree/fixErrorMsg) you can find the fix branch.

valegagge, 2025-01-07:
Hi @AntonioViscomi I guess we can close thi issue.
What do you think?

AntonioViscomi, 2025-01-07:
> Hi [@AntonioViscomi](https://github.com/AntonioViscomi) I guess we can close thi issue. What do you think?

Hi @valegagge ,
I think we can close the issue

closed 2025-01-08 by AntonioViscomi