robotology/icub-firmware #554: Current set to zero after switching the control mode to current

GiulioRomualdi, 2025-01-13:
I’m currently working on the current control mode and noticed an issue. When switching the control mode from position to current, the setpoint is set to 0.

However, when switching from position to torque, the setpoint in torque is set to match the estimated value.

After discussing this with @valegagge, it seems this might be a bug in the 2foc implementation, where the setpoint in the current mode is not being set correctly.

Here, it's a video showing the behaviour 

https://github.com/user-attachments/assets/c1a9356a-9321-414f-91fd-5f50b53d91e2

GiulioRomualdi, 2025-01-14:
To provide a high-level perspective: The problem with this behavior is that when we switch to current mode while the robot is on the ground, the desired current is momentarily set to zero. Although this happens for a very short time, it is sufficient for the robot to collapse under its own weight.

GiulioRomualdi, 2025-01-14:
Cc @isorrentino @giulero @loremoretti

Nicogene, 2025-01-14:
Ccing @ale-git that maybe knows what is happening

valegagge, 2025-01-16:
I provided the ems fw here attached to @GiulioRomualdi to test on the robot. 
Currently, the fw contains only a debug message to understand in detail what happening in the current control mode switching. 

[image]

Please note the the fw I provided is based on the current version on the robot (3.95) and not on the top of the devel branch.

Thanks @GiulioRomualdi for the tests!

[ems_x_giulio_v_3_195.zip](https://github.com/user-attachments/files/18441345/ems_x_giulio_v_3_195.zip)

GiulioRomualdi, 2025-01-16:
I'm currently testing the firmware on ergoCubSN002. I flashed the board 10.0.1.6 the right upper leg

[image]

Here the output of the interface
[interface_current.txt](https://github.com/user-attachments/files/18441908/interface_current.txt)

This is the video of the experiment

https://github.com/user-attachments/assets/94967be4-3f65-4e5d-a271-a7927c523d17

valegagge, 2025-01-16:
Hi @GiulioRomualdi ,
is it clear from your test that the references provided to the pid are 0 as we supposed before.

Now the solution is to provide the current value of `current`, but... we need to filter it in some way, because its values change a lot!

GiulioRomualdi, 2025-01-16:
> Now the solution is to provide the current value of current, but... we need to filter it in some way, because its values change a lot!

A possibility is to to use the same filter @ale-git implemented back in time (an average between n samples discarding the max / min values)

valegagge, 2025-01-16:
> A possibility is to to use the same filter [@ale-git](https://github.com/ale-git) implemented back in time (an average between n samples discarding the max / min values)

yep!
I'm checking exactly that!

ale-git, 2025-01-17:
Since the 2FOC can run in three different control modes (pwm, current, velocity), we decided for the clean transition mode that resets the references until a new reference value is received. We can use instead the smooth transition mode, that uses the present feedback as new reference. Thus, 1) moving from current or pwm mode to velocity, we can set the measured velocity as reference, 2) moving from current or velocity to pwm, we can set the present pwm as reference, 3) moving from velocity or pwm to current, we can use the present current feedback (filtered) as input.

valegagge, 2025-01-17:
Hi @ale-git ,
I'm implementing the point 3. as soon as I complete it I'll ask you to review.

In addition, we need to remember that the fw of 2foc 3.3.38 already provides a filtered current.

ale-git, 2025-01-17:
Ok. Just for clarity, my proposal is to implement all points 1, 2 and 3.

valegagge, 2025-01-17:
> Ok. Just for clarity, my proposal is to implement all points 1, 2 and 3.

sure! I see that in some cases there are some smooth transitions, but I need to go in detail with all of them.

In order to fix this issue I started from the point 3.

valegagge, 2025-01-31:
Hi guys,
Today I tested on a bench setup the fix. 

As you can see from the following video currently when we switch from the `position `to `current `the reference provided is 0 and therefore the joint moves due to the weights. (It behaves like when is in `idle`.) 

https://github.com/user-attachments/assets/36e150ce-af32-4219-8a51-760cd335945b

I fixed this behavior and now on switching in `current` mode, the fw sets the average of the current. 
The average is calculated by a very simple implementation of  first-order moving average.

As you can see from the following video, I tested the fix on the bench setup at it seems to work as expected. 
Is important to notice that the current show in the `yarpmotorgui` doesn't correspond to the one printed on the state ext port. (To make more readable the value of the current, I splitted the stateExt port and read on the part related to the current). 

https://github.com/user-attachments/assets/763d8bfd-bc5e-416e-ac1e-34ae0d5b1097

I dumped some current samples and the calculated average and this is the plot:

[image].

Now we need to test on the robot.

The code is here https://github.com/valegagge/icub-firmware/tree/FixCurrModeOnDevel and it is aligned with the devel branch.

valegagge, 2025-02-14:
@GiulioRomualdi told me that the fw on the robot doesn't work. so booked the robot Monday morning to debug the problem.

GiulioRomualdi, 2025-02-14:
My fault, today I retested the firmware that I flashed on the left leg.  

In the following video, I show the **yarpmotorgui** and the robot. From the **motorgui**, you can see that we first set the **left leg hip pitch** in **current mode**, and then we do the same for the right leg. In both cases, the motor current drops to 0 (**@valegagge** already mentioned this to me, and it seems to be a known bug—perhaps we should open an issue somewhere).  

https://github.com/user-attachments/assets/a171c13a-ca72-4004-9e29-e8530d1d6693

In this other video, we can see how the switch affects the robot's behavior. On the board with the fix, switching from **position control** to **current control** keeps the leg still (which is the intended behavior). However, in the other case, the leg moves unexpectedly.  

https://github.com/user-attachments/assets/64104365-73fa-44f2-be0f-040fabd314ab

In this last video, we observe the measured current. Unfortunately, we do not have the desired setpoint. However, for the **left leg**, the switch is **imperceptible**, while for the **right leg** (running the old firmware), the desired current is automatically set to 0.  

https://github.com/user-attachments/assets/e784513c-3bfa-42fc-be6e-59683d9a81f1

---

Dataset: [robot_logger_device_2025_02_14_13_02_10.zip](https://github.com/user-attachments/files/18799781/robot_logger_device_2025_02_14_13_02_10.zip)

valegagge, 2025-02-17:
Hi @GiulioRomualdi ,
thanks for your report. I am waiting for the meeting scheduled on Wednesday 19th, to discuss the results.

valegagge, 2025-03-21:
PR merged. I'm closing this issue since the required behavior has been implemented and works as expected.

closed 2025-03-21 by valegagge