robotology/icub-firmware #556: Improve the AMO encoder diagnostic to make the user aware of specific AMO encoder problem

MSECode, 2025-01-15:
Following the works done in https://github.com/robotology/icub-firmware/issues/535 and collected in this EPIC: https://github.com/icub-tech-iit/study-encoders/issues/21 we have decided to exploit the information made available by the diagnostic bytes of the AMO encoder, i.e. the ones called `status_0` and `status_1`, to make the user more aware of what is actually happing with the encoder when the problems taken into account during the previous issues happen.

The diagnostic is fully described by these 2 tables available in the AMO datasheet.

[image]

MSECode, 2025-01-15:
I've started the work at this branch: https://github.com/MSECode/icub-firmware/commits/feature/updateAMOEncDiagnostic/

@valegagge, I've a doubt about using the embot library to check which bits of the diagnostic byte are set to 1. However, in order to do so we need to include `embot_core_binary.h` in `EOappEncodersReader.c`. Do u think that this might be an issue or is it ok?
Moreover, what is now just missing is:
- adding the specific errors to the error class: `eoerror_category_HardWare` and decide which we wanna consider as blocking error and which as just warning
- identify thresholds for stopping the joint if the number of diagnostic fails increases too much
- update the diagnostic in icub-main
- decide if we wanna add some fields to this enum `hal_spiencoder_diagnostic_type_t` or create a new one for the specific AMO failures

valegagge, 2025-01-15:
Hi @MSECode,

regarding the including of `embot_core_binary.h` at first check it seems ok, but i don't know if it creates a inclusion loop.

> adding the specific errors to the error class: eoerror_category_HardWare and decide which we wanna consider as blocking error and which as just warning

Here, since you carried on the AMO study, you could suggest something.

> identify thresholds for stopping the joint if the number of diagnostic fails increases too much

Currently, if there is a very critical case, I'd like to add a fault.

> decide if we wanna add some fields to this enum hal_spiencoder_diagnostic_type_t or create a new one for the specific AMO failures

You can suggest a solution that best fits the fw and icub-main if such errors are shared with the diagnostic at high level.

MSECode, 2025-01-20:
Now, just to test if we can correctly read the different diagnostic bits I've added some debug lines, which are sent to the yarprobointerface every 10 seconds if we have errors. Doing so, I've run a test with the joint with the threeEncoders that was moving between -180 and +180 degrees in position mode using the AMO encoder only calibrated with calibration type 12.
At a certain point I've moved the encoder boards slightly far from the magnetic disk and this is a part of the output I got:

```
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=535s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[DEBUG] from BOARD 10.0.1.2 (three-encoders-setup) time=535s 209m 77u :  src LOCAL, adr 0,(code 0x04000003, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag03  Poor level or clipping in both master and nonius
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=536s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=537s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[ERROR] from BOARD 10.0.1.2 (three-encoders-setup) time=537s 396m 146u :  MC: hard limit reached. The joint position is outside its hardware boundaries. (Joint=three_encoders_joint (NIB=0))
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=538s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=539s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=540s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=541s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=542s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[ERROR] from BOARD 10.0.1.2 (three-encoders-setup) time=542s 397m 144u :  MC: hard limit reached. The joint position is outside its hardware boundaries. (Joint=three_encoders_joint (NIB=0))
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=543s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[ERROR] positionMoveRaw: skipping command because  BOARD three-encoders-setup (IP 10.0.1.2)   joint  0  is not in VOCAB_CM_POSITION mode
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=544s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=545s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[DEBUG] from BOARD 10.0.1.2 (three-encoders-setup) time=545s 210m 73u :  src LOCAL, adr 0,(code 0x04000007, par16 0x0000 par64 0x0000000000000008) -> DEBUG: tag07  All error bits are risen in the par64 value
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=546s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=547s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[ERROR] from BOARD 10.0.1.2 (three-encoders-setup) time=547s 398m 144u :  MC: hard limit reached. The joint position is outside its hardware boundaries. (Joint=three_encoders_joint (NIB=0))
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=548s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[... cut ...]
[WARNING] from BOARD 10.0.1.2 (three-encoders-setup) time=553s 158m 92u :  MC: Absolute encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=three_encoders_joint (NIB=0), encoderPort=0, encoderType=AMO)
[ERROR] positionMoveRaw: skipping command because  BOARD three-encoders-setup (IP 10.0.1.2)   joint  0  is not in VOCAB_CM_POSITION mode
```

Among the `WARNING` messages you can see the `DEBUG` messages we used for checking if the bit of the diagnostic are correctly captured.

MSECode, 2025-01-20:
Navigating the code, I've found that a similar work was previously done with these functions: https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/embobj/plus/board/EOappEncodersReader.c#L1984-L2143

However, I suppose that the code in those lines it is not currently used since I think that can be enabled by the flag that we have in icub-main: `eomn_serv_diagn_mode_MC_AMOyarp == mcdiagnostics.config.mode`. 
Anyway, I was thinking to add the part of my code that rise different errors depending on the values in the registries `status 0` and status 1` to those functions.
I thought to that since those methods are already well organized and managed. On top of that we can or remove the flag in icub-main or re-enable the AMO diagnostic.
What do u think @valegagge ? Does it make sense to you or should I proceed differently.
Once that is done, I just need to add the error and warning messages and the threshold but the part that reads the bits of the registry values is almost complete.

MSECode, 2025-01-27:
After a discussion w/ @valegagge we have chosen to define the code similarly to what done for the aksim2 encoder.
Therefore I've updated the code at this branch: https://github.com/MSECode/icub-firmware/tree/feature/updateAMOEncDiagnostic
and tested it.
I've noticed that when using the AMO in our test setup, either calibrating the joint with calibration type 10 or 12, we tend to have quite a lot of errors correlated to the bit `NON_CTR` shown in the `status 1` diagnostic. During normal use, instead, the other bits of the diagnostic registries remain to zero.
Considering this, we are gonna do some other test after a fine calibration of AMO encoder and then we will add here some plots and data.

MSECode, 2025-01-29:
After the modifications to the code I did some tests for:
1. Checking that the errors are sent to the `yarprobotinterface` 
2. Checking if before and after a fine calibration of the encoder the number of errors changes

Here the results:
1. Regarding point one we can see from the section of the logs below that the errors are correctly sent to the upper level. Regarding the errors shown I should say that when the joint is running (these logs come from a try made before the fine calibration of the encoder) we have lots of errors related to the period inconsistency. Instead, for triggering all the other errors I moved the encoder board far from the magnetic disk.

[image]

2. Then for the second point we can see how, after a fine calibration of the AMO encoder, the number of errors basically disappeared:

**Before --> lots of period inconsistency errors**

[image]

**After --> no errors anymore**

[image]

valegagge, 2025-02-21:
PR merged. Can we close?

MSECode, 2025-02-21:
Sure, PR linked is done.

closed 2025-02-21 by MSECode