robotology/icub-firmware #573: Setup firmware toolchain

traversaro, 2025-03-06:
Hi everyone! I am trying to setup the firmware dev toolchain, I am following the documentation until I reached the point https://icub-tech-iit.github.io/documentation/icub_firmware/fw_toolchain/installing_keil/#configuring-icub-firmware .

I work at IIT, but it does not seem that `\\storage05.icub.iit.local\repository\software\ipal-extra` is still available. Perhaps the same file is available in a different location? Thanks!

traversaro, 2025-03-06:
Just to clarify, I tought I had to follow that step otherwise the ems compilation failed with:

~~~
Build started: Project: ems4rd.diagnostic2ready
*** Using Compiler 'V6.23', folder: 'C:\Users\straversaro\AppData\Local\Keil_v5\ARM\ARMCLANG\Bin'
Build target 'ems-ulpro'
linking...
.\Obj\ems004.axf: error: L6002U: Could not open file ..\..\..\..\..\libs\highlevel\abslayer\ipal\lib\ipal.base.medium.cm4.lib: No such file or directory
Not enough information to list image symbols.
Not enough information to list load addresses in the image map.
Finished: 2 information, 0 warning, 0 error and 1 fatal error messages.
".\Obj\ems004.axf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:01
~~~

traversaro, 2025-03-06:
Compiling  as suggested by @SanLordKevin results in error:

~~~
Build started: Project: ipal
*** Target 'ipal.base.medium.cm4' uses ARM-Compiler 'Default Compiler Version 5' which is not available.
*** Please review the installed ARM Compiler Versions:
   'Manage Project Items - Folders/Extensions' to manage ARM Compiler Versions.
   'Options for Target - Target' to select an ARM Compiler Version for the target.
*** Build aborted.
Build Time Elapsed:  00:00:00
~~~

I will try to follow the error indication.

SanLordKevin, 2025-03-06:
I posted here a quick guide to install the compiler version 5:
- https://github.com/icub-tech-iit/knowledge-base/discussions/24

traversaro, 2025-03-09:
Thanks a lot @SanLordKevin ! Unfortunately that documentation is private, perhaps it could be moved somewhere public and linked in https://icub-tech-iit.github.io/documentation/icub_firmware/fw_toolchain/installing_keil/#configuring-icub-firmware ?

Anyhow, I followed your instructions and now when I try to compile the `emBODY/eBcode/arch-arm/libs/highlevel/abslayer/ipal/ipal.uvoptx` project the compilation fails with:

~~~
Build started: Project: ipal
*** Using Compiler 'V5.06 update 7 (build 960)', folder: 'C:\Users\straversaro\AppData\Local\Keil_v5\ARM\Arm_Compiler_5.06u\Bin'
Build target 'ipal.base.medium.cm4'
compiling ipal_arp.c...
src\tcpnet\ipal_arp.c(42): error:  #5: cannot open source input file "ipiit.h": No such file or directory
  #include "ipiit.h"
src\tcpnet\ipal_arp.c: 0 warnings, 1 error
compiling ipal_eth.c...
src\tcpnet\ipal_eth.c(42): error:  #5: cannot open source input file "ipiit.h": No such file or directory
  #include "ipiit.h"
src\tcpnet\ipal_eth.c: 0 warnings, 1 error
compiling ipal_ipiit_storage.c...
"no source": Error:  #5: cannot open source input file "ipal-extra\ipal_ipiit_storage.c": No such file or directory
ipal-extra\ipal_ipiit_storage.c: 0 warnings, 1 error
compiling ipal_igmp.c...
src\tcpnet\ipal_igmp.c(42): error:  #5: cannot open source input file "ipiit.h": No such file or directory
  #include "ipiit.h"
src\tcpnet\ipal_igmp.c: 0 warnings, 1 error
compiling ipal_base.c...
src\tcpnet\ipal_base.c(42): error:  #5: cannot open source input file "ipiit.h": No such file or directory
  #include "ipiit.h"
src\tcpnet\ipal_base.c: 0 warnings, 1 error
compiling ipal_udp.c...
src\tcpnet\ipal_udp.c(41): error:  #5: cannot open source input file "ipiit.h": No such file or directory
  #include "ipiit.h"
src\tcpnet\ipal_udp.c: 0 warnings, 1 error
compiling ipal_sys.c...
src\tcpnet\ipal_sys.c(42): error:  #5: cannot open source input file "ipiit.h": No such file or directory
  #include "ipiit.h"
src\tcpnet\ipal_sys.c: 0 warnings, 1 error
".\Obj\ipal_base_medium_cm4.lib" - 7 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:00
~~~

I looked for the `ipiit.h` file in the repo or on internet, but I could not find it.

traversaro, 2025-03-10:
> I looked for the `ipiit.h` file in the repo or on internet, but I could not find it.

Thanks to @marcoaccame that pointed to another internal doc, I was able to compile the ipac lib, and now the compilation works fine, while the linking fails with:

~~~
Build started: Project: mc4plus.diagnostic2ready
*** Using Compiler 'V6.23', folder: 'C:\Users\straversaro\AppData\Local\Keil_v5\ARM\ARMCLANG\Bin'
Build target 'mc4plus-ulpro'
linking...
.\Obj\mc4plus.axf: error: L6050U: The code size of this image (261162 bytes) exceeds the maximum allowed for this version of the linker.
Finished: 0 information, 0 warning, 0 error and 1 fatal error messages.
".\Obj\mc4plus.axf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:02
~~~

I am not sure if that is expected, as I installed the KEIL MDK community as indicated in https://icub-tech-iit.github.io/documentation/icub_firmware/fw_toolchain/installing_keil/#download-keil .

SanLordKevin, 2025-07-25:
Sorry for the delay on the answer, for me no problem, but personally I don't know where to put this information.