robotology/icub-firmware #577: RX phase of the control loop last more than wanted

GiulioRomualdi, 2025-03-12:
When starting the an application tha streams setpoints at 1khz the board of the legs starts streaming several warnings expplaining that the RX phase of the control loop lasts more then expected

```
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=654s 800m 791u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 362[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=654s 808m 302u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 307[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
```

As can be seen by the commit https://github.com/icub-tech-iit/robots-configuration/commit/0d633a4f54c984a41a24cee37347ab4d85284480 the robot configuration files are set in a way the ControlBoard_nws_yarp streams data at 1khz and the same is done by the embObjMotionControl

This is the version of the firmware on the robot

[image]

GiulioRomualdi, 2025-03-12:
I know that we can manually tune this parameters https://github.com/icub-tech-iit/robots-configuration/blob/0d633a4f54c984a41a24cee37347ab4d85284480/ergoCubSN000/hardware/electronics/right_leg-eb6-j0_3-eln.xml#L18-L26 
but I was wondering if there exist a way to solve the problem by optimize some bottleneck in the firmware

GiulioRomualdi, 2025-03-21:
This is actually what is happening when testing the walking on the ergoCubSN001 

https://github.com/user-attachments/assets/8f5984ea-8b40-497e-abfb-015c1be3773c

is there anyone that can help me in this?

pattacini, 2025-03-21:
cc @marcoaccame

marcoaccame, 2025-03-21:
Having said that ...

## 1. why maybe RX phase takes so long

The reason why the RX phase takes slightly longer is maybe because these two boards manage four joints each and I assume that messages with setpoints to those joints are transmitted very frequently (maybe at 1 kHz). 
also, the diagnostics emission takes some time so it makes things even worst.

## 2. how to effectively measure what is the best timing for rx-do-tx

In this [PR](https://github.com/robotology/icub-main/pull/971#issue-2392384075) I added a feature in xml files so that one can turn on statistics about min-average-max execution time of every rx-do-tx. 

So, if you run with these additional settings

```xml
<group name="LOGGING">

    <!-- Contains enabling / disabling of logging that must be emitted immediately as soon as the condition appears --> 
    <group name="IMMEDIATE">
        <!-- enable / disable the emission of diagnostics related to the measured execution time of RX higher than the configured
             maxTimeOfRXactivity and similarly for DO and tx
          -->
        <param name="emitRXDOTXoverflow">       false </param>
    </group>
    
    <!-- Contains enabling / disabling of logging that must be emitted periodically -->
    <group name="PERIODIC">
        <!-- the period expressed in seconds in the interval [1.0, 600.0] -->
        <param name="period">                   10          </param>
        <!-- enable / disable the periodic emission of diagnostics about to the statistics of teh execution time of the RX, DO and TX phase
             in the form of minimul, average and maximum values.
          -->                        
        <param name="emitRXDOTXstatistics">     true        </param>
    </group>

</group>

```

you can have a clear indication of the values to assign to `maxTimeOfRXactivity`, `maxTimeOfDoactivity`, `maxTimeOfTXactivity`.

Then you should restore as follow

```xml
<group name="LOGGING">

    <group name="IMMEDIATE">
        <param name="emitRXDOTXoverflow">       true </param>
    </group>
    
    <group name="PERIODIC">
        <param name="period">                   10          </param>                   
        <param name="emitRXDOTXstatistics">     false </param>
    </group>

</group>

```

## 3. how to disable the annoying diagnostics about rx-do-tx timing overflow

I must say that if a timing overflow happens then something is wrong because the real time behavior is nit satisfied and that we must (should) never disable these warnings.

BUT if after running step 2 and having measured that the sum of the max (or even worst average) execution times in rx-do-tx is > 1000 then ..... we could just add this section.

```xml
<group name="LOGGING">

    <group name="IMMEDIATE">
        <param name="emitRXDOTXoverflow">       false </param>
    </group>
    
    <group name="PERIODIC">
        <param name="period">                   10          </param>                   
        <param name="emitRXDOTXstatistics">     false </param>
    </group>

</group>

```

marcoaccame, 2025-03-21:
P.S. I just noticed that the above info is not in this [file](https://github.com/robotology/robots-configuration/blob/master/iCubTemplates/iCubTemplateV6_0/hardware/electronics/body_part--ebX-jA_B-eln.xml), so I will fill it it.

traversaro, 2025-05-02:
Using the dissector added in https://github.com/robotology/icub-firmware-shared/pull/111 and a pcap dump collected by @isorrentino  (see [walking_ergocub001_1khz_v1_42.zip](https://github.com/user-attachments/files/20009958/walking_ergocub001_1khz_v1_42.zip)) I think I got more insight on this issue.

I do not how frequently the "SYS: the RX phase of the control loop has last more than wanted. RX execution time 362[usec]" is printed during the experiment in which @isorrentino collected the dataset, but my assumption is that there are some UDP packets that for some reason are bigger then usual, and then the RX phase take more.

So, during normal operation the UDP packet sent from the ergocub-torso to the board (the log looks at the traffic to the `10.0.1.6`) contains 8 ROPs:

[image]

expanding the ROPs, you can see the 4 are position setpoints, and 4 are the torque feedback:

[image]

As the walking is for now position controlled, one possible optimization is not send the torque feedback if it is not needed (i.e. if the axis is not torque controller), but as eventually we may want the system to work fine even if we do low level torque control, probably that is not a super-critical implementation.

However, this is the "normal" operation, but are there moment in which more ROPs are sent, for some reason? We can easily filter for those cases with the `ip.dst == 10.0.1.6 and embobj_rop.nrops > 8` filter. That revealed a lot of packets in which 12 ROP were sent, and by inspecting them, it seems that apparently either a duplicated setpoint or a duplicated torque feedback was sent, twice for the same index. There were even cases in which 16 ROPs were sent:

[image]

In that case, both the torque feedback and setpoints are duplicated.

My hypothesis is that, given that the whole machinery of how the position setpoints are produced, i.e. : 

setpoints: walking-controller (controller process) --> YARP nws (controller process, thread 2) --> YARP nwc (yarprobotinterface process) --> embObjMotionControl (yarprobotinterface process)
torque-feedback: wholebodydynamics (yarprobotinterface process) --> embObjMotionControl (yarprobotinterface process)

as this periodic threads are not synchronized, it may happen that in the same embObjMotionControl slot, either the walking controller or wholebodydynamics (or both, as in the case of NROPs == 16) provide two times the position setpoint or the torque feedback. In that case, I guess it could be a reasonable optimization to ensure that if `embObjMotionControl` receives two times the position setpoint or the torque feedback, only the last one is actually sent to the board.

traversaro, 2025-05-02:
The relevant piece of code are:
* For the update of the torque feedback: https://github.com/robotology/icub-main/blob/8961dcf869d1e7cc3bafed47b2fab2fe014fca71/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4092-L4109
* For the position setpoint: https://github.com/robotology/icub-main/blob/8961dcf869d1e7cc3bafed47b2fab2fe014fca71/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4466-L4489

traversaro, 2025-05-02:
Call stack for both code paths:

* [`EthResource::setRemoteValue`](https://github.com/robotology/icub-main/blob/1c91cf2c368002e95c1f471d77d46c24f958c343/src/libraries/icubmod/embObjLib/ethResource.cpp#L774)
* [`eth::theNVmanager::set`](https://github.com/robotology/icub-main/blob/1c91cf2c368002e95c1f471d77d46c24f958c343/src/libraries/icubmod/embObjLib/theNVmanager.cpp#L1186)
* [`eth::theNVmanager::Impl::set`](https://github.com/robotology/icub-main/blob/1c91cf2c368002e95c1f471d77d46c24f958c343/src/libraries/icubmod/embObjLib/theNVmanager.cpp#L995)
* [`HostTransceiver::addROPset`](https://github.com/robotology/icub-main/blob/8961dcf869d1e7cc3bafed47b2fab2fe014fca71/src/libraries/icubmod/embObjLib/hostTransceiver.cpp#L402)
* [`HostTransceiver::addSetROP__`](https://github.com/robotology/icub-main/blob/8961dcf869d1e7cc3bafed47b2fab2fe014fca71/src/libraries/icubmod/embObjLib/hostTransceiver.cpp#L292)
* [`eo_transceiver_OccasionalROP_Load`](https://github.com/robotology/icub-firmware-shared/blob/774a09375b10ae3b2fbdad51f4d87f5dc539672e/eth/embobj/plus/comm-v2/transport/EOtransceiver.c#L530)
* [`eo_transmitter_occasional_rops_Load`](https://github.com/robotology/icub-firmware-shared/blob/f4a60f5e18d9bd42a590c76edfc2a8a9f2debf44/eth/embobj/plus/comm-v2/transport/EOtransmitter.c#L1276C19-L1276C54)
* [`s_eo_transmitter_rops_Load`](https://github.com/robotology/icub-firmware-shared/blob/f4a60f5e18d9bd42a590c76edfc2a8a9f2debf44/eth/embobj/plus/comm-v2/transport/EOtransmitter.c#L1419)
* [`eo_ropframe_ROP_Add`](https://github.com/robotology/icub-firmware-shared/blob/f4a60f5e18d9bd42a590c76edfc2a8a9f2debf44/eth/embobj/plus/comm-v2/transport/EOropframe.c#L473)

At a first glance, the optimization is not obvious as the deserialization as soon as a SET ROP is added, and so it is not obvious once a new SET ROP is added to go back to the existing ROP to modify it with the new value.

traversaro, 2025-05-02:
@GiulioRomualdi @isorrentino when you have time it would be cool to test https://github.com/robotology/icub-firmware-shared/pull/112 .

traversaro, 2025-05-02:
The question I kept me asking today is: even if indeed the duplicated send of setpoint and estimated torques can be mitigated, why it happens so frequently? Even if the threads are not synchronized and the OS does not use realtime facilities, that was still a bit surprising. So, I plotted arrival time of to/from packets from the board, and I compare it against an ideal rate of 1 packet every 1 ms, and this is the result:

[image]

for some reason, the packets send from the board are quite regular, while the packages send to the board in a short time (~100 ms) they reach a drift of 10 ms. The fact that the rate is not regular may be related to the use of a non-realtime grade OS, but the fact that the drift is always in the same direction may suggest that something could be improved in how the thread that sends the data is scheduled.

The data for reproducing the plot can be found at  [plot_data.zip](https://github.com/user-attachments/files/20018111/plot_data.zip).

traversaro, 2025-05-02:
The problem seems exactly what is described in https://github.com/robotology/yarp/issues/2488, and indeed the `ethSender` thread uses `yarp::os::PeriodicThread` to implement a periodic thread (see https://github.com/robotology/icub-main/blob/1c91cf2c368002e95c1f471d77d46c24f958c343/src/libraries/icubmod/embObjLib/ethSender.cpp#L62). We could probably try to exploit the solution found in https://github.com/robotology/yarp/issues/2488 also in the ethSender/ethReceiver code.

traversaro, 2025-05-02:
I think the drift problem may be fixed by https://github.com/robotology/icub-main/pull/1018 . 

To recap, it would be interesting to see if the warning problem is mitigated by either https://github.com/robotology/icub-main/pull/1018 or https://github.com/robotology/icub-firmware-shared/pull/112, but I am much more confident of https://github.com/robotology/icub-main/pull/1018  being correct, while for https://github.com/robotology/icub-firmware-shared/pull/112 I am not really sure everything is implemented correctly.

isorrentino, 2025-05-03:
@traversaro, sure, I can test it. I'll collect the tcpdump with https://github.com/robotology/icub-main/pull/1018 and https://github.com/robotology/icub-firmware-shared/pull/112 separately, and then with the two modifications together.

traversaro, 2025-05-03:
> @traversaro, sure, I can test it. I'll collect the tcpdump with https://github.com/robotology/icub-main/pull/1018 and https://github.com/robotology/icub-firmware-shared/pull/112 separately, and then with the two modifications together.

Thanks! It would be great also to collect (at least qualitatively) if the warnings are still there!