robotology/icub-firmware #603: Great current control mismatch between desired and feedback

GiulioRomualdi, 2025-05-27:
We tried to control the joint of the right hip pitch in current mode. To do this we designed a simple application that send the command via YARP https://github.com/ami-iit/element_joint-velocity-control/tree/450bff2fedbf6a7cf920ea8aa494e806572c980b/scripts. The leg suddenly moved

The following plot show the desired current and the measured one. In theory they should be the same.

[image]

This is the version of the firmware

[image]

The firmware updater doesn't give me the possibility to switch the board in maintenance so I cannot see the version of the 2foc

In the meantime @fils99 and @loremoretti are checking al dataset to see when the current control stopped working

> [!IMPORTANT]
> From the first inspection of the data we noticed that the bug affects both SN000 and SN001

fils99, 2025-05-27:
In a dataset collected on May 9th, I see:

[image]

So it was NOT working fine on May 9th.

Application used: https://github.com/fils99/bipedal-locomotion-framework/commit/a4e2100aaef01f70edbf65c5ff3415b553464b1a

GiulioRomualdi, 2025-05-27:
> In a dataset collected on May 9th, I see:
> 
> [image]
> 
> So it was NOT working fine on May 9th.
> 
> Application used: [fils99/bipedal-locomotion-framework@a4e2100](https://github.com/fils99/bipedal-locomotion-framework/commit/a4e2100aaef01f70edbf65c5ff3415b553464b1a)

Bingo! This explain the cause of this issue: https://github.com/robotology/icub-tech-support/issues/2125

With @LoreMoretti we started the current control and the hip suddenly moved to the limit breaking the robot

ale-git, 2025-05-27:
I think it's essential to read the 2FOC firmware version at this point... I remember that there should be a way to see the 2FOC boards by connecting directly to the ethernet on a leg. 

What about the other joint controlled by the same 2FOC board? Did you try to run the same test on it?

maggia80, 2025-05-27:
@ale-git Could you help check the FW and do the tests together?

ale-git, 2025-05-27:
- The 3.3.42 version increases the maximum PWM to 96% from 80%.
- It seems to me that the power supply has been increased to 48V. In my knowledge, 48V is the limit for the 2FOC board. I'd try to reduce it a little, about 45V.

|Mar 18 | Mar 21 | Apr 3 |
|--------|---------|--------| 
| 3.3.39 | 3.3.40| 3.3.42 |

[image]

valegagge, 2025-05-27:
Hi @ale-git ,
During the demo of [Google Mind (23th April)](https://github.com/ami-iit/lab-events-demos/issues/359) the current control works fine. 
As you can see [here ](https://github.com/robotology/icub-tech-support/issues/2114#issuecomment-2801719023) on ergocub-sn001, the fw has been updated with 2foc version 3.3.42.

This means that version 3.3.42 works as expected. This is consistent with the data shown in the [comment above. ](https://github.com/robotology/icub-firmware/issues/603#issuecomment-2911974345).

So we can conclude that the only fw of 2foc with the changes of PWM limit is not the main cause of the issue.

We need to understand what happened after the 29th April.

ale-git, 2025-05-28:
I've flashed the devel 3.3.42 fw version from repo, and everything works fine now.
I moved the joints in current control from the yarpmotorgui and they moved as expected.
There were in home several test version of 2FOC firmware, I deleted them.

LoreMoretti, 2025-05-28:
The I tried again with `right_hip_pitch`:

https://github.com/user-attachments/assets/07e2d000-8a53-46d7-8930-96ca09d5bc6c

With the experiment log:

[image]

It does not seem to work for `r_hip_pitch`.

GiulioRomualdi, 2025-05-28:
We deeply analyzed the behaviour and we discovered that https://github.com/ami-iit/bipedal-locomotion-framework/pull/964 introduces a breaking change in the system. 
It applies to the torque, a scaling factor (180/pi), even if the joint was controlled in current. 
Still, it is not clear to me why it was working fine for the ankle but not for the hip.

Given the importance of this, I will revert immediately the commit in blf. Release the software and write a message in the teams.

traversaro, 2025-05-28:
Can't we just reintroduce the if :

~~~
            // if the control mode is torque or PWM current it is not required to change the unit of 
             // measurement 
             if (mode == ControlMode::Torque // 
                 || mode == ControlMode::PWM // 
                 || mode == ControlMode::Current)
~~~ 

that were removed in the PR that introduced the regression?

GiulioRomualdi, 2025-05-28:
I'm scared that the ankle (of Ergocub) was working properly while the hip was not