robotology/icub-firmware #605: Validate the developed Benf

valegagge, 2025-05-27:
We developed the Back EMF on 2foc. See here for more info: https://github.com/icub-tech-iit/tickets/issues/3978

In addition this new feature has been used during the PID tuning on the linear actuator of the ankle. See here: https://github.com/ami-iit/component_joint-control/issues/8#issuecomment-2909868285.

We want to validate the feature in order to release it.

**DoD**
- define a validation tests list
- perform the defined test
- prepare the documentation
- prepare PRs

ale-git, 2025-05-28:
@isorrentino is colleting a large amout of data from several experiments (sine waves, square waves, step, ramps, ...), with and without payload.

valegagge, 2025-05-30:
On 28 May, @ale-git, @isorrentino and @pattacini and I have a meeting regarding the improvements of 2foc fw that @ale-git developed and that were necessary as the results of tests performed on ankle testbench pointed out. (see [here](https://github.com/ami-iit/component_joint-control/issues/8#issuecomment-2909868285))

We decided to make "official" all the 4 improvements already developed and test on the ankle testbench.

We agreed to make another test campaign to validate these changes also on the motor and ergocub 2.0 joint.

ale-git, 2025-06-03:
### Improvement 2 - Feed Forward terms

At present, the closed loop current control implementation is:

$$
V_{pwm} = K_P(I_{ref}(t)-I_{meas}(t)) + K_I \sum_{k=0}^t(I_{ref}(k)-I_{meas}(k))
$$

We can improve the performance of the controller by using some knowledge a priori about the physics of the electric motor and its parameters:

$$
V = RI + K_e\omega + L\frac{dI}{dt}
$$

We neglect the term that involves the current derivative because the current measure is very noisy, but the other two terms can be introduced in the control loop as feed forward terms:

$$
V_{pwm} = K_P(I_{ref}(t)-I_{meas}(t)) + K_I \sum_{k=0}^t(I_{ref}(k)-I_{meas}(k)) + K_{FF}I + K_{BEMF}\omega
$$

https://github.com/ale-git/icub-firmware/blob/c1819b1f7b3bce3e642b0a2185e672e29722cbfe/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/2FOC.c#L905

ale-git, 2025-06-13:
Yesterday we succesfully tested the FW on Ergocub 0000. We tested both normal operation (position, ecc) and direct current control withount unexpected behaviour.

ale-git, 2025-06-19:
Hi @valegagge, some test results here https://github.com/ami-iit/component_joint-control/issues/8

valegagge, 2025-06-20:
> Hi [@valegagge](https://github.com/valegagge), some test results here [ami-iit/component_joint-control#8](https://github.com/ami-iit/component_joint-control/issues/8)

Thanks, @ale-git ! I saw that result, but in my previous message (sorry if I wasn't clearer!), I was actually talking about the test results from the robot itself. When you say we haven't seen any unexpected behavior in normal operation and current control, what exactly do you mean by that? And do you think just a visual check is enough for us here?

valegagge, 2025-06-30:
@ale-git @isorrentino: Can we consider the test phase concluded?

cc @pattacini

isorrentino, 2025-07-01:
Hi @ale-git @valegagge do you think we can proceed with the PR?

ale-git, 2025-07-01:
> Hi [@ale-git](https://github.com/ale-git) [@valegagge](https://github.com/valegagge) do you think we can proceed with the PR?

Hi @isorrentino, I've made the PR.

valegagge, 2025-07-03:
PR : https://github.com/robotology/icub-firmware/pull/618