robotology/icub-firmware #614: Increazing ROP size

valegagge, 2025-06-17:
Do we need to increase the ROP size? see https://github.com/robotology/icub-firmware/pull/586#issuecomment-2971140706.

valegagge, 2025-06-17:
# Meeting Report: Evaluation of ROP Size Increase

**Date:** June 17, 2025
**Subject:** Discussion on the necessity and methods for increasing the size of the ROP (Run-time Object Protocol) within UDP packets.
**Involved people** @marcoaccame @ale-git 

---

### Introduction
The meeting's objective was to assess the need to **increase the ROP size from 384 bytes to 392 bytes** to accommodate new configuration parameters.

### Current Problem
The current ROP size is **384 bytes**. A need has arisen to include new configuration data, specifically the **timeout values for various control modes** and the **torque feedback timeout**.

### Proposed Solutions and Discussion

1.  **Inclusion of New Data in the `cmd_setpoint` Structure:**
    * **Proposal:** Initially, it was suggested to include the new timeouts directly in the `cmd_setpoint` structure.
    * **Decision:** This solution was **rejected**. The main reason is related to the **watchdog** logic, which must activate immediately upon a change in control mode. Including timeouts in `cmd_setpoint` would not allow for this desired reactivity.

2.  **Inclusion of Timeouts in the `set_control_mode` Command:**
    * **Proposal:** The second solution discussed involved including the setpoint timeout and feedback timeout information directly within the `set_control_mode` command.
    * **Implications:** This solution would entail **significant implementation changes**, requiring alterations to the protocol, non-volatile variables (NVs), and callbacks.

---

### Considerations for Increasing ROP Size

It was noted that the parameters discussed today (timeouts) are likely **not the last to be added** to the configuration. Soon, further additions might be necessary, such as parameters for "Benf" and "velocity direct." This led to considering **increasing the ROP size** as a more forward-thinking solution.

However, **significant difficulties** emerged regarding this option:

1.  **"Magic Number" of 384 Bytes:** The current value of 384 bytes is **widespread and hardcoded across multiple sections of the codebase**, in both `icub-main` and the firmware (fw). This makes a simple change to the ROP size a complex operation with a high risk of errors.

2.  **MTU (Maximum Transmission Unit) Issues:**
    * Even with the current size of 384 bytes, it's **not possible to send 4 ROP configurations in the same UDP packet** without exceeding the standard MTU of 1500 bytes.
    * It was highlighted that **`embObjMotionControl` sends one joint configuration at a time** and also performs a check (sets the configuration, requests it back, and verifies data matching). This process, while robust, underscores the current limitation.

3.  **Architectural Definition:** At an architectural level, it is crucial that the **ROP size is defined at a single point** and **shared between the firmware and `icub-main`** to ensure consistency and facilitate future modifications.

---

### Further Proposal
An additional proposal was made to **split the configuration into two distinct structures**. However, this solution would also require a **modification of the existing protocol**.

### Next Steps
A **new meeting has been scheduled for tomorrow** to continue the discussion and delve deeper into the proposed solutions, evaluating their long-term implications.

ale-git, 2025-06-18:
I have increased the ROP size in icub-main and icub-firmware-shared (the code is in ale-git repos, devel branch). Tomorrow I'll test it

marcoaccame, 2025-07-04:
Hi @ale-git, see my comment in here: https://github.com/robotology/icub-firmware/pull/586#issuecomment-3035967148

and the following PRs that should close the issue:
- https://github.com/robotology/icub-firmware/pull/620
- https://github.com/robotology/icub-firmware-shared/pull/116