robotology/icub-firmware #615: Velocity control improvements

valegagge, 2025-06-20:
After different types of experiments have been out that we need to increase the usability of the Velocity control.

We started we a a meeting where we decided the type of velocity control we can provide.

In this epic, we want to keep track of all the work necessary to complete these improvements. 

Please refer to the following comment with the MoM of the meeting where we defined the high-level goals.

valegagge, 2025-06-20:
# Enhancement: Implement New Velocity Control Options and Refine Configuration

## Description

This issue outlines the planned improvements to our existing velocity control strategies, focusing on expanding functionality and streamlining configuration. Our current system primarily relies on **minJerk** for input velocities.

---

## New Velocity Control Options

We've decided to integrate two additional velocity control methods:

### 1. Velocity with PWM Output
- **Status**: Tested on *ergoCub sn001*.
- **Documentation**: Detailed results are available in the relevant repository.

### 2. Velocity with Internal Current Loop
- **Status**: Tested on the ankle joint.
- **Note**: During these tests, the *minJerk* calculation on the EMS (EtherCAT Master Slave) was not removed.

The original velocity control method will now be referred to as **"Classic"**.

---

## Implementation Strategy

To ensure these three control types are readily available and manageable, we will:

- Utilize configuration parameters to allow selection between the different control modes.
- Continue to use the existing `IVelocityControl` interface.

---

## Action Items

Here are the next steps identified for this enhancement:

- ✅ Complete the documentation for the BEMF tests.
- 🔍 Investigate `IVelocityControl` for any missing functionalities.
  - If additional features are required, explore alternative solutions to avoid direct modification of the interface to accelerate development.
- 📅 Schedule a dedicated technical meeting to determine the optimal approach for implementing the new parameters.
  - Ensure compatibility with both legacy code and the new AMC-family boards.

---

randaz81, 2025-06-27:
> Add support for sending both position and velocity references simultaneously, or assess whether this can be achieved using existing interfaces available in YARP. This allows using velocity as feedforward when operating in cascaded Position → Velocity → PWM or Position → Velocity → Current control modes.

On the YARP side, a control mode called MIXED is already available. While a detailed documentation about what control law it actually implements, in the past it was used exactly for this purpose, i.e. POS+VEL

isorrentino, 2025-06-27:
Thank you for the suggestion. I think the current implementation of the MIXED control mode differs somewhat from the improvements we would like to have. If you find it helpful, we can schedule a brief meeting to align on this.

isorrentino, 2025-07-24:
## Meeting Minutes
23th July 2025

### Attendees

@valegagge, @randaz81, @ale-git, @marcoaccame, @traversaro, @Nicogene, @isorrentino 

### Discussion Summary

#### Velocity Control Improvements

We started discussing improvements to velocity control. It was agreed to move forward with adding a new control mode called **Velocity Direct**.

### Technical Points Clarified

* **PID Interface**: currently allows setting PID parameters using an enum.
* The **POSITION PID** is applied across position, position direct, and velocity modes.
* PID interfaces exist in configuration parsers but are missing from YARP. For instance, the parser already supports velocity direct control (@valegagge correct me if I got it wrong).
* The original velocity control mode will remain unchanged, and the new control mode will be explicitly named **Velocity Direct Control**.

### Agreed Changes

* New YARP control mode: **Velocity Direct**.
* A new interface for Velocity Direct will include only `setVelocity`, propagated in NWC, NWS, and remapper.
* Configuration files will clearly distinguish:

  * Standard velocity control (existing method).
  * Velocity Direct Control (with explicit gains, defaulting speedPID to velocity direct).
* YARP’s `setPID` methods will now accept enum options: `position`, `velocitydirect`.

### Mixed Mode Clarifications
The Mixed mode is currently operational only for eye movements.

We can set both position and velocity through dedicated interfaces.
We discuss whether to maintain only one Mixed mode, and its specific behavior will depend on the definitions set in the robot configuration files.

### Future Considerations

* @randaz81 suggested, for future iterations, to move away from using enums for PID settings and instead define explicit `setPID` and `getPID` methods for each control mode.

### Action Points

* [ ] Implement the new interface for the Velocity Direct control mode. @randaz81 
* [ ] Provide example configuration files covering all control modes. @WhoEverWantsToContribute
* [ ] Write documentation for the templates in `robots-configuration`. @ale-git 

Here is the repo for writing the motor control configuration files
https://github.com/isorrentino/motor-control-configuration

ale-git, 2025-07-25:
I've written the documentation about EMS/2FOC control mode combinations here:

https://github.com/icub-tech-iit/five/issues/339

ale-git, 2025-07-25:
This is the configuration with a `MinJerk` mode (Position/Velocity/Mixed) and `current` output mode:

[image]

ale-git, 2025-07-25:
This is the configuration with a `MinJerk` mode (Position/Velocity/Mixed) and `velocity` output mode:

[image]

ale-git, 2025-07-25:
This is the configuration with a `MinJerk` mode (Position/Velocity/Mixed) and `velocity_current` output mode.
**The `velocity_current` output mode is already implemented in the 2FOC board but not implemented in the configuration protocol (to be added).**

[image]

ale-git, 2025-07-25:
This is the configuration with a `position Direct` mode and `PWM` output mode.
**A PID dedicated to Direct mode is already implemented in the EMS board but not implemented in the configuration protocol (to be added).**

[image]

ale-git, 2025-07-25:
This is the configuration with a `position Direct` mode and `velocity` output mode.
**A PID dedicated to Direct mode is already implemented in the EMS board but not implemented in the configuration protocol (to be added).**

[image]

ale-git, 2025-07-25:
This is the configuration with a `position Direct` mode and `velocity_current` output mode.
**A PID dedicated to `Direct` mode is already implemented in the EMS board but not implemented in the configuration protocol (to be added).
The `velocity_current` output mode is already implemented in the 2FOC board but not implemented in the configuration protocol (to be added).**
[image]

ale-git, 2025-07-25:
### REMARK

The Ipeak current limiter that makes use of `currentPid` parameters is alway present in 2FOC control modes when an explicit current control loop isn't included in control (pwm, velocity). It is omitted in 2FOC velocity modes for clarity.

ale-git, 2025-07-25:
Aren't the images visible now?

isorrentino, 2025-07-25:
I believe the documentation we're referring to is similar to what I found at this link: https://github.com/robotology/robots-configuration/blob/master/iCubTemplates/iCubTemplateV6_0/hardware/motorControl/body_part--ebX-jA_B-mc.xml, along with all the possible control modes currently implemented that you described in the comments above.

ale-git, 2025-07-25:
The documentation that you posted above is still perfectly valid. The schemes above are an integration that allows to understand at a glance what runs on what board in each control mode combination, and what's the name of the parameter set involved in the configuration files, and which parameters configure the particular control mode combination.

ale-git, 2025-08-07:
This control schema would provide a flexible structure for all possible position/velocity control modes with stiff interaction.
The advantage is that we don't have to implement a schema for each combination, but we can taylor our control law at .xml configuration level by simply setting some constants to 0 in order to cut out some branches. Also, we can taylor the control law depending on the available feedback and sensor resolutions.

The trajectory model can be minjerk, trapezoidal, direct, and so on.

[image]

isorrentino, 2025-08-11:
You should be able to access it now!

valegagge, 2025-09-02:
Here is the proposed configuration file for the motion control during the meeting on [12th August](https://github.com/robotology/icub-firmware/issues/615#issuecomment-3182335038). 

[newFileCFG.xml](https://github.com/user-attachments/files/22094511/newFileCFG.xml)

The main changes are:
- introduction of veldir parameter in `CONTROLS` group
- add the inputType parameter in PID configuration 
- Use the `ControlLaw` param to specify if we want to use the trajectory generator or not (in the future, we can specify which trajectory ).

valegagge, 2025-09-22:
# Update Meeting: 17/09/2025

**Participants:**  
@isorrentino , @randaz81 , @valegagge , @ale-git , @marcoaccame , @AntonioViscomi   

---

### Discussion
Ines pointed out that **`speedPid`** and **`velocityDirect`** are redundant, since both are parameters used to characterize the velocity PID loop closed on the motor.  
The working group decided to remove the `speedPid` parameter in favor of the new `velocityDirect`, as the velocity PID will always run on the motor side.  

---

### Update on Action Points from [Previous Meeting](https://github.com/robotology/icub-firmware/issues/615#issuecomment-3278925814)
- Add tabs for Position Direct and Velocity Direct in yarpmotorgui → @randaz81 (TODO)  
- Parser modifications for the new XML structure → @valegagge, Antonio Viscomi (TODO)  
- Meeting on low-level velocity control logic → @isorrentino, @valegagge, @ale-git ✔️  
- Passport renewal → @randaz81 (TODO)  
- Finalize and test new CAN message → @marcoaccame, @ale-git (ONGOING)  
- Double-check the possibility of sending feedforward in position control (potentially already available in PID interface) → @randaz81: `setPidFeedForward` added in `IPidControl` ✔️  

---

### New Action Points (including unfinished items from previous meeting)
- Add tabs for Position Direct and Velocity Direct in yarpmotorgui → @randaz81  
- Parser modifications for the new XML structure → @valegagge, Antonio Viscomi  
- Passport renewal → @randaz81  
- Finalize and test new CAN message → @marcoaccame, @ale-git (ONGOING)  
- Fix documentation in YARP: `PositionDirectRaw` is present, but `PositionDirect` is missing → @randaz81  
- Add tabs for Position Direct and Velocity Direct in yarpmotorgui → @randaz81  
- Fix documentation in YARP: `PositionDirectRaw` is present, but `PositionDirect` is missing → @randaz81  

---

### Steps Required to Enable the Velocity Direct Feature
**Note:** YARP 4.0 must be used.  

1. Make devices compilable and aligned with YARP 4.0 (provide empty implementations for the new methods added to interfaces).  
2. Update the complete pipeline from device to firmware:  
   - Modify the parser  
   - Modify the protocol to send the new information (if necessary)  
   - Update `embobjMotionControl` to handle the new configuration  
   - Properly implement the new methods  
   - Modify the firmware to implement `velocityDirect`  

------------------------------

Please add or modify details if I misunderstood something.