robotology/icub-firmware #627: Test the amc fw of PR icub-firmware-build#218

valegagge, 2025-07-28:
This test is related to the timeouts feature added by the fw PR: https://github.com/robotology/icub-firmware/pull/586.

Unfortunately, in the last test phase we forgot to test the amc board since it shares the embedded motion control.

In this task, we want to validate the feature on the amc board.

**DoD**
We verified that with the new fw in this PR  icub-firmware-build#218:
1. if we set the PWM control mode it works as expected, unless we disconnect the Ethernet cable and the controller sets the joint in hw fault with the error message related to the timeout expires. In addition we need to check that the PWM provides to the motor or the current is 0. (We can achieve this by reading the stateExt port)
2. same as point 1, but in current mode
3. same as point 1, but in velocity mode.

The results have been reported here.

AntonioViscomi, 2025-07-30:
@valegagge and I tested the firmware on both the AMC board and the EMS + 2FOC combo, focusing on timeout behavior in PWM mode.

### EMS Results:
Timeouts appear to be functioning correctly:[^1]
|EMS: 10 [msec] (OK) |EMS: 100 [msec] (OK)|
|---	|---	|
| https://github.com/user-attachments/assets/21f07cdd-0d5a-489b-a43f-f8e0d65b22fe | https://github.com/user-attachments/assets/bb365d8a-67a6-4a3c-9555-bdf2d6cc441a |

### AMC Results
Timeouts are not behaving as expected:

|AMC: 0 [msec] (OK) | AMC: 100 [msec] (NOT OK)|
|---	|---	|
|https://github.com/user-attachments/assets/7f5e9504-d2b9-4646-94aa-8b4d13434c0f|https://github.com/user-attachments/assets/7d398871-5df7-4f15-97c4-118aba9de953|

Given that yarpmotorgui sends reference commands every 50 ms, a 100 ms timeout should allow for normal operation. However, when switching to PWM mode with a 100 ms timeout, the system immediately enters hardware fault mode.

[^1]: For the EMS + 2FOC testing we had to use the @ale-git setup since we experienced issues with the torso pitch setup.

ale-git, 2025-08-04:
@valegagge since the `mc` is shared between AMC and EMS boards, it is strange that they show different behaviours.
In my understanding, the send command timer in the yarpmotorgui is enabled only if "enable openloop control" is set.
Are you sure it was enabled when you did the test?

MSECode, 2025-08-06:
Hi,
from the comments above I see that  the joint went in pwm without enabling it. This should not happen, should it. So there might be some issue with the latest pr in the motorgui or I'm mistaken.  
Basically if you do not enable the mode you should not be able to move the slider. It should be grayed out. 

cc: @AntonioViscomi @valegagge

valegagge, 2025-08-11:
> After correct configuration, the AMC is running and the timeout response is the same of EMS board as expected. The only difference is that a larger timeout is required (300 ms) to avoid the timeout the first time that the board is set to PWM or current mode. It seems like there is a lag in the loop yarpmotorgui -> yarprobotinterface -> AMC [@marcoaccame](https://github.com/marcoaccame) [@SanLordKevin](https://github.com/SanLordKevin)

Good catch, @ale-git !

To make this feature work, while we are waiting for the study of lag you noticed, we need to update the default value of the timeout. 
First of all, I noticed in your PR that you are using the value `100` instaed of the parameter `defaultVelocityTimeout` in the function `bool Parser::parseTimeoutsGroup(yarp::os::Searchable &config, std::vector<timeouts_t> &timeouts, int defaultVelocityTimeout)`.

Moreover, we can use that parameter to set the default to 100 for EMS and 300 for AMC until the lag issue is fixed. If there is any. (cc @marcoaccame @SanLordKevin )

ale-git, 2025-08-11:
@valegagge with the last commit I've set the default timeouts to 300 ms for AMC and 100 ms for the other boards.