robotology/icub-firmware #637: Review enhancement kff and kbemf

valegagge, 2025-09-02:
Following the development carried on in #621, we need to review the changes. 

We need to discuss the format of the new CAN messages introduced for configuring the new parameter with the team 4 (@marcoaccame ).

marcoaccame, 2025-09-03:
Now we have the following CAN messages that set the KP, KI, KD, K0, limit on output and limit on integral for instance for the current PID.

```mermaid
---
config: 
  packet: 
    rowHeight: 32
    bitWidth: 128
    bitsPerRow: 8
    showBits: true
---
packet-beta
title SET_CURRENT_PID
  0-0: "OPC = 0x65"
  1-2: "KP [int16_t]"
  3-4: "KI [int16_t]"
  5-6: "KD [int16_t]"
  7-7: "KS [uint8_t]"
  
```
**Figure**. The actual payload of the **ICUBCANPROTO_POL_MC_CMD__SET_CURRENT_PID** message. It is 8 bytes long.

```mermaid
---
config: 
  packet: 
    rowHeight: 32
    bitWidth: 128
    bitsPerRow: 8
    showBits: true
---
packet-beta
title SET_CURRENT_PIDLIMITS
  0-0: "OPC = 0x67"
  1-2: "K0 [int16_t]"
  3-4: "LIMOUT [int16_t]"
  5-6: "LIMINT [int16_t]"
  
```
**Figure**. The actual payload of the **ICUBCANPROTO_POL_MC_CMD__SET_CURRENT_PIDLIMITS** message. It is 7 bytes long.

And we also have the relevant get messages. For a full reference see xxx.

We already have more PID parameters than what the CAN messages can handle

```c++
typedef struct
{
    float32_t               kp;                 /**< proportional gain */
    float32_t               ki;                 /**< integral gain */
    float32_t               kd;                 /**< derivative gain */
    float32_t               kff;
    float32_t               limitonintegral;    /**< limit of integral term */
    float32_t               limitonoutput;      /**< limit of the output of the pid */
    float32_t               offset;             /**< the k0 in the pid formula */
    float32_t               stiction_up_val;
    float32_t               stiction_down_val;
    int8_t                  scale;
    uint8_t                 filler[3];
} eOmc_PIDfl_t;             EO_VERIFYsizeof(eOmc_PIDfl_t, 40)
```

And from this [PR](https://github.com/robotology/icub-firmware-shared/pull/119), we shall also have a `float32_t Kbemf` inside `eOmc_motor_config_t`

So, we surely need a way to express at least `kff` of a PID and possibly all the parameters in `float32_t` 

So, the proposal is  this: use a new additional SET_PID message that is able to transport any possible parameters in `float32_t`  in the following way.

```mermaid
---
config: 
  packet: 
    rowHeight: 32
    bitWidth: 128
    bitsPerRow: 8
    showBits: true
---
packet-beta
title SET_PID
  0-0: "OPC = 0x7D"
  1-1: "TYPE [PIDtype]"
  2-2: "PARAM [PIDparam]"
  3-3: "FFU"
  4-7: "VALUE [float32_t]"
  
```
**Figure**. The payload of the **ICUBCANPROTO_POL_MC_CMD__SET_PID** message. It is 8 bytes long.

Where:

```C++

enum class PIDtype : uint8_t 
{ 
    CURR = 0, VEL = 1, POS = 2, // and others if needed
    NONE = 15 
};

enum class PIDparam : uint8_t 
{ 
    KP = 0, KI = 1, KD = 2, 
    KFF = 3, 
    LIMINT = 4, LIMOUT = 5,
    STICTION_UP = 6, STICTION_DOWN = 7, // maybe useful 
    KBEMF = 8, // possibly even if it is a motor parameter. TO BE DECIDED!!! 
    LOCK = 13, UNLOCK = 14, NONE = 15 
};
```

A possible simple usage is the following

```mermaid
sequenceDiagram
    actor HOST
    actor Motor Board
    
    
    HOST->>Motor Board: SET_PID<PIDtype::VEL, PIDparam::KFF, 10.333>
    activate Motor Board
    Note right of Motor Board: imposes kff of velocity PID in usage to be = 10.333
    deactivate Motor Board
```

**Figure**. The messages **ICUBCANPROTO_POL_MC_CMD__SET_PID** used to perform the change of KFF of the velocity  PID without protection of other values.

Whereas a more advanced use is this one

```mermaid
sequenceDiagram
    actor HOST
    actor Motor Board
    HOST->>Motor Board: SET_PID<PIDtype::CURR, PIDparam::LOCK, 0>
    activate Motor Board
    Note right of Motor Board: gets a copy of the current PID to be CPID
    HOST->>Motor Board: SET_PID<PIDtype::CURR, PIDparam::KP, 13.333>
    Note right of Motor Board: sets CPID.kp = 13.333
    HOST->>Motor Board: SET_PID<PIDtype::CURR, PIDparam::KI, 2.1>
    Note right of Motor Board: sets CPID.ki = 2.1
    HOST->>Motor Board: SET_PID<PIDtype::CURR, PIDparam::KFF, 1.1>
    Note right of Motor Board: sets CPID.kff = 1.1
    HOST->>Motor Board: SET_PID<PIDtype::CURR, PIDparam::UNLOCK, 0>
    deactivate Motor Board
    Note right of Motor Board: copy CPID int the current PID to be used

```

**Figure**. The messages **ICUBCANPROTO_POL_MC_CMD__SET_PID** used to perform an atomic change of KP and KI and KFF of the current  PID.

marcoaccame, 2025-09-03:
If we use the above  `ICUBCANPROTO_POL_MC_CMD__SET_PID` message, then the `Motor.c` would emit the `kff` in this way:

```mermaid
sequenceDiagram
    actor Motor.c
    actor 2foc

    Motor.c->>2foc: SET_VELOCITY_PID<KP=3, KI=4, KD=5, scale=0>
    activate 2foc
    Note right of 2foc: imposes {kp, ki, kd} of the velocity PID to use {3, 4, 5}
    deactivate 2foc
    
    Motor.c->>2foc: SET_PID<PIDtype::VEL, PIDparam::KFF, 10.333>
    activate 2foc
    Note right of 2foc: imposes kff of velocity PID in usage to be = 10.333
    deactivate 2foc
```

What about the `kbemf`? If it is OK to consider kbemf a PID parameter, at reception of the motor config we could just send a

```mermaid
sequenceDiagram
    actor Motor.c
    actor 2foc

    Motor.c->>2foc: SET_MOTOR_CONFIG<...>
    activate 2foc
    Note right of 2foc: accepts what inside the message
    deactivate 2foc
    
    Motor.c->>2foc: SET_PID<PIDtype::CUR, PIDparam::KBEMF, 3.3>
    activate 2foc
    Note right of 2foc: imposes kbemf of current PID in usage to be = 3.3
    deactivate 2foc
```

If we dont like that the kbemf is considered a PID value we can introduce a message `SET_MOTOR_CONFIG2` that contains the `kbemf` in float32_t format, maybe using the same strategy as `SET_PID` uses.

marcoaccame, 2025-09-04:
### The `kbemf` transported over  a `SET_PID`

If it is OK to consider `kbemf` a PID parameter, at reception of the motor config from UDP, the `Motor.c` we could just send:

```mermaid
sequenceDiagram
    actor Motor.c
    actor 2foc

    Motor.c->>2foc: SET_MOTOR_CONFIG<...>
    activate 2foc
    Note right of 2foc: accepts what inside the message
    deactivate 2foc
    
    Motor.c->>2foc: SET_PID<PIDtype::CUR, PIDparam::KBEMF, 3.3>
    activate 2foc
    Note right of 2foc: imposes kbemf of current PID in usage to be = 3.3
    deactivate 2foc
```

However, it may be worth to add `kbemf` inside a new `SET_MOTOR_CONFIG2` also with other params such as `Vmax`,  motor resistance and inductance, control-specific params etc.

A proposal is in next post.

pattacini, 2025-09-04:
Nice "visual" doc!

valegagge, 2025-09-08:
I added @ale-git as an assignee because of the changes required in the previous comments.

marcoaccame, 2025-09-09:
> I added [@ale-git](https://github.com/ale-git) as an assignee because of the changes required in the previous comments.

OK, I will get in touch w/ @ale-git so that we can proceed w/ the changes

marcoaccame, 2025-09-24:
> Hi [@marcoaccame](https://github.com/marcoaccame) and [@valegagge](https://github.com/valegagge) , I've debugged the 2FOC receiver part and now all is working. I've updated the PR accordingly.

good, so we are ready to start the merging. pls have a look also at gemini's comment in [here](https://github.com/robotology/robots-configuration/pull/752#pullrequestreview-3172858422).

valegagge, 2025-10-01:
PR merged! closing

closed 2025-10-01 by valegagge